MotorolaMobilityLLC/readme #5: How to use

SHA89, 2015-09-30:
I'm trying to make a build for the surnia. I was following the README file, and for building the kernel, all was fine.
However, for building the whole android, i'm confused. The README for surnia told me that i should get following git projects:

bionic
external/clang
external/dnsmasq
external/esd
external/gcc-demangle
external/iproute2
external/iptables
external/libcap-ng
kernel
system/core
vendor/qcom/opensource/wlan/prima

So i did, (accept for esd, i was unable to find this folder).

But when i was looking at the remaining projects of MotorolaMobilityLLC, i noticed that there were also other projects with a release for the surnia.

Do i need these projects too? 
If i download all the projects with a surnia release, and unzip them in the aosp source, should the build be succesful? Meaning, using all the surnia releases, i should be able to make a working ROM? or are there sometimes still bugs in it, causing the build to fail?

Kind regards,
Stijn

tags07, 2015-09-30:
We only publish the repos that are copyleft or those linking with GPLv2 components. You should be able to build target binaries from these sources and swap the corresponding binaries on the device. This is not the complete ROM. We DO NOT publish components that have proprietary software or anything we are not obligated to post sources.

SHA89, 2015-09-30:
ok, thanks for your quick reply.
And should i only use the repos that are mentioned in the corresponding README?

RoneyThomas, 2015-10-03:
Could someone post the updated build instructions for Surnia android 5.1.1 (MMI-LPI23.29-23). The 5.0.2 build instruction for Sunria says that we should replace external/esd with Motorola provided version.  But it's not provided by Motorola. Is it because of the copyright issue???

tags07, 2015-10-05:
@RoneyThomas  external/esd should not be in that list. We never use that repo in our production releases. We'll post an updated build instruction for MMI-LPI23.29-23 soon.

RoneyThomas, 2016-03-22:
@tags07 Now that the kernal sources for android 6.0 is published for surnia. Could you please post the updated build instruction for MMI-MPI24.65-39 please.

charleseb, 2016-03-22:
Hi Roney,

I'm working to update and release the build instructions related to such
version. It will be released soon.

Thanks,
Charles

RoneyThomas, 2016-04-12:
@charleseb so i got the marshmallow update today for my moto e(xt1527). Is there an update for build instructions???

charleseb, 2016-04-12:
Hi Roney,

We are still working to release this read me file.
I'm facing a problem with one module on the process and I think we'll
release that instructions in a few days.

Thanks,
Charles

charleseb, 2016-04-12:
Hi Roney,

I just published the readme file to this version.

https://github.com/MotorolaMobilityLLC/readme/commit/15a326bf7412d444c7df4863b84833788ad9ca71

Thanks,
Charles

closed 2019-04-05 by charleseb