MotorolaMobilityLLC/readme #8: Moto g 2014 programmer files

Thecrazyskull, 2015-11-26:
This is probably not the best place to ask, but is motorola (if yes, when?) planning to release the programmer files compatible with the bootloader shipped in the 6.0 soak test?
The programmer files which are currently available online are 4.4.4 ones and are not compatible with the current bootloader, thus the people who are stuck in qhsusb dload mode due to accidently downgrading their bootloader because being simply unaware of the fact that the soak test updates the bootloader (or just blind me) are unable to get out of it.
I hope motorola would release these files.
Thanks.

Thecrazyskull, 2015-11-28:
@tags07 @ccfries

Thecrazyskull, 2015-11-30:
I reply would be nice you know... @tags07 @ccfries

ccfries, 2015-11-30:
What mechanism did you use to do the downgrade to 4.4.4?  Fastboot will
block going backwards to an older bootloader when the security version is
updated (without bricking the phone).  Is this TWRP or some other custom
recovery?

Thecrazyskull, 2015-11-30:
I installed the xt1069 factory image (was neccesary for the soak test), then i installed the 6.0 soak test over it.
I had some issues with my modem on the 6.0 soak test so i reverted back to the latest xt1068 factory image (the one with the stagefright fixes) via fastboot.
So i did it via fastboot, not with any custom recovery.

tags07, 2015-11-30:
Do you have a secure device or a DE/Pure Android device?

Thecrazyskull, 2015-11-30:
@tags07 a regular secure device.

Thecrazyskull, 2015-12-02:
@tags07 @ccfries any update on this?

Thecrazyskull, 2015-12-03:
Just sent my device for repair, hope they will accept it :)

closed 2015-12-26 by Thecrazyskull