MotorolaMobilityLLC/readme #10: qcacld-2.0 issue

melihercan, 2016-01-06:
Hi,

Moto X Style, xt1572, Lollipop5.1,1: 
I have followed the steps specified in this readme:
https://github.com/MotorolaMobilityLLC/readme/blob/master/MMI-LPH23.116-18.1.txt
to build my own custom kernel and WLAN driver. 
The only difference is, I used the single config file msm8994-perf_defconfig and skipped the extra 2 debug configs specified in the readme. The final .config file exactly matches the stock .config. 

WARNING: the kernel built with master branch of
git clone https://android.googlesource.com/platform/prebuilts/gcc/linux-x86/aarch64/aarch64-linux-android-4.9
FAILS. See the discussion here, I observed exactly the same symptoms:
https://github.com/MotorolaMobilityLLC/kernel-msm/issues/38
USE marhmallow_release branch.

I extracted the boot.img from phone and unpack it to a folder. I replaced the kernel with my Image and repack the boot.img. I flash the new boot image and everything is OK. Even the stock WLAN driver is OK. So I guess my kernel matches the stock one.

Then I built the qcacld-2.0 WLAN driver (lollipop branch). The first thing that surprises me is the wlan.ko file size, it is approx 97MB!!! Something is fishy here. Anyway I copied the wlan.ko to a new folder on /system/lib/modules and changed the soft link wlan.ko pointing my wlan.ko. Disabled WiFi, rmmod the stock wlan.ko and insmod my wlan.ko. Of course "Module hash not found in whitelist" problem. I added a new config with single line:
# CONFIG_MODULE_WHITELIST is not set
and the final .config CONFIG_MODULE_WHITELIST=y is replaced with the above line. 

With the new kernel, stock WLAN is still OK. I disabled WiFi and insmod my Wlan.ko. This time the driver is loaded OK and produced exactly the same debug messages as stock one. But I couldn't connect to the AP. So I rebooted the phone with wlan.ko soft link pointing to my driver. But no WiFi then!!! I can't enable the WiFi button. I checked the driver and found out that it was not loaded!!! I manually loaded it but WiFi button still stuck in off. I rebooted again and checked dmesg. For some reason my wlan.ko is not loaded during startup. Loading it after system startup manually doesn't help  
enabling of WiFi. So it seems the WiFi stack is expecting a running driver started during boot.
Switched back to stock driver and this driver is loaded during boot and WiFi is OK.

So something is preventing my custom wlan driver to be loaded during boot, something other than whitelist mechanism. Cause I don't see any attempts loading it on dmesg. I wonder if it is the size ~97MB. But when I manually load it and check lsmod it shows ~5MB image similar to stock one. 

Any idea what might be going wrong? And why am I getting such a huge size wlan.ko?

Thanks,

melihercan, 2016-01-06:
Update: 
I stripped the 97 MB wlan.ko with 
aarch64-linux-android-strip –g wlan.ko
to get rid of the debug information and size is back to ~5MB like the stock one. When manually installed, it works. BUT still the module is NOT loaded during the boot!!! There should be some sort of protection there. I made a search on ram disk scripts but couldn't find anything. Any help/hint would be appreciated.  

Thanks

vinimed, 2021-05-06:
Ticket is too old. If you are still interested on request, please re-open the issue.

closed 2021-05-06 by vinimed