MotorolaMobilityLLC/readme #12: How to compile "v4l2-hal.ko" ?

erfanoabdi, 2016-11-08:
Hi, 
i built my custom kernel for Moto Z (griffin) and everything looks good i have all modules except "v4l2-hal.ko" i know it's related to Camera Mod (Hasselblad) and "Service 'mods_camd'" but i don't know how to compile this module 
i searched web about "v4l2-hal.c" and nothing founds 
my dmesg without it : 
`init: Service 'mods_camd' (pid 1057) exited with status 0`
`init: Service 'mods_camd' (pid 1057) killing any children in process group`

please help me to compile this 
thanks

ccfries, 2016-11-08:
let me check, there should be a motorola/external/v4l2_hal

erfanoabdi, 2016-11-09:
sorry but I can't find it could you please send link 
Thanks

closed 2016-11-11 by erfanoabdi

ccfries, 2016-11-11:
Yeah, sorry we left it out, it was a mistake in how we added the license files in this new git and the tools missed it.  Charles just pushed it up.

-Chris