MotorolaMobilityLLC/readme #25: Build instructions for Moto G7 Plus (lake)

Jleeblanch, 2019-04-30:
@charleseb 

Could you please post the kernel building info for the Moto G7 Plus (lake) for tag PPW29.98-28? 

I can successfully compile a kernel using the available sources, however I've yet to be able to get it to boot, and it's not compiling the kernel modules located at /vendor/lib/modules (which I see you created a repo (found here: https://github.com/MotorolaMobilityLLC/motorola-kernel-modules) but it seems to only have tag PPO29.80-27 currently, although those seem to be the modules (or at least some of them) located at /vendor/lib/modules. Im sure there is something I'm missing like the correct DTC version (or prebuilt?) so if you could post that information I'd greatly appreciate it. 

Thank you,
Joshua

alexenferman, 2020-12-08:
Same thing here with the Moto E 2020 (ginna). WLAN drivers are an issue in my case.

Jleeblanch, 2020-12-17:
I'm closing this as we've seen releases for lake on all necessary fronts and have no issues building the kernel and modules. @alexenferman if you still need help, I'd recommend opening a separate issue for ginna.

closed 2020-12-17 by Jleeblanch

alexenferman, 2020-12-17:
I found the fix as well, I used sdm632 kernel which was not from Moto and includes all the required modules