MotorolaMobilityLLC/readme #26: instruction request for nora kernel builds.

amaterasushen, 2019-10-04:
There are no instructions for compiling the Motorola E5 Nora. The only instruction it has is for the Motorola E5 Plus: MMI-OPP27.91-113 and for the Motorola E5 Play: MMI-OPG28.54-19.

the instructions are very distinct from each other. The Moto E5 nora font released so far are MMI-OPP27.91-176-8 and MMI-OPP27.91-146

chenyt9, 2019-10-29:
@amaterasushen,
The build instructions have been updated. Please refer to this file:
https://github.com/MotorolaMobilityLLC/readme/commit/7026676866ce803f282d2ada26227398c6672a2a
Thanks.

closed 2019-10-29 by chenyt9

amaterasushen, 2019-10-29:
> @amaterasushen,
> The build instructions have been updated. Please refer to this file:
> [7026676](https://github.com/MotorolaMobilityLLC/readme/commit/7026676866ce803f282d2ada26227398c6672a2a)
> Thanks.

Thank you noble man.