MotorolaMobilityLLC/readme #27: Build Instructions for Moto G7 Play (Channel)

keyur2maru, 2020-03-12:
@charleseb can you please release build guide for channel?

Also while trying to compile for channel, I'm getting this error if you can take a look
CC drivers/video/fbdev/msm/mdss_mdp_pp_debug.o
../drivers/media/radio/radio-iris.c: In function 'iris_vidioc_s_ctrl':
../drivers/media/radio/radio-iris.c:4516:6: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
error, forbidden warning: radio-iris.c:4516
../scripts/Makefile.build:339: recipe for target 'drivers/media/radio/radio-iris.o' failed
make[4]: *** [drivers/media/radio/radio-iris.o] Error 1
../scripts/Makefile.build:653: recipe for target 'drivers/media/radio' failed
make[3]: *** [drivers/media/radio] Error 2

keyur2maru, 2020-03-13:
@charleseb I was able to obtain the Image file and boot it up, I also pushed all the obtained modules to /system/vendor/lib/modules/ 
but still for some reason touchscreen isn't working
can you possibly help?

awheeler-mmi, 2020-03-13:
@keyur2maru Anything in dmesg? I believe that channel is using the ilitek_mmi touchscreen driver -- can you make sure it's loaded? Also, make sure the firmware is in /vendor/firmware.

keyur2maru, 2020-03-13:
@awheeler-mmi thank you for reply. I cannot check dmesg because I am on stock rom so no adb permissions as I cannot use touch.. 
My make command was 
make -j64 -C O=out/ DTC_EXT=dtc CONFIG_BUILD_ARM64_DT_OVERLAY=y ARCH=arm64 CROSS_COMPILE=$across KCFLAGS=-mno-android CONFIG_INPUT_FOCALTECH_0FLASH_MMI_IC_NAME=ft8719 CONFIG_INPUT_FOCALTECH_0FLASH_MMI_IC_NAME=ft8719 CONFIG_INPUT_HIMAX_V2_MMI_IC_NAME=hx83102 CONFIG_INPUT_HIMAX_V2_MMI_IC_NAME=hx83112

If it ilitek_mmi touchscreen driver, then my config is wrong right?

awheeler-mmi, 2020-03-13:
Looks like channel could have one of three touch panels: himax, novatek, or ilitek.    See sdm632-channel-common-overlay.dtsi.

keyur2maru, 2020-03-13:
@awheeler-mmi Yes, in the sdm632-channel-common-overlay.dtsi I can see three touch panels. Does that means I can use any one of them?

keyur2maru, 2020-03-13:
`channel:/vendor/firmware # ls
FT8006U_Pramboot.bin      cpp_firmware_v1_12_0.fw                        
a225_pfp.fw               cpp_firmware_v1_1_1.fw                         
a225_pm4.fw               cpp_firmware_v1_1_6.fw                         
a225p5_pm4.fw             cpp_firmware_v1_2_0.fw                         
a300_pfp.fw               cpp_firmware_v1_4_0.fw                         
a300_pm4.fw               cpp_firmware_v1_5_0.fw                         
a330_pfp.fw               cpp_firmware_v1_5_1.fw                         
a330_pm4.fw               cpp_firmware_v1_5_2.fw                         
a420_pfp.fw               cpp_firmware_v1_6_0.fw                         
a420_pm4.fw               cpp_firmware_v1_8_0.fw                         
a530_gpmu.fw2             emmc_ffu_13_S0J9N8.fw                          
a530_pfp.fw               emmc_ffu_15_FE12MB.fw                          
a530_pm4.fw               emmc_ffu_15_QE13MB.fw                          
a530v1_pfp.fw             emmc_ffu_15_RC14MB.fw                          
a530v1_pm4.fw             emmc_ffu_15_RC1BMB.fw                          
a530v2_seq.fw2            emmc_ffu_15_RE1BMB.fw                          
a530v3_gpmu.fw2           emmc_ffu_15_RX14MB.fw                          
a530v3_seq.fw2            emmc_ffu_15_RX1BMB.fw                          
a540_gpmu.fw2             ilitek-dsbj-ILI9881-0B-6C-channel.bin          
aw8624_haptic.bin         leia_pfp_470.fw                                
aw8624_rtp.bin            leia_pm4_470.fw                                
aw8624_rtp_lighthouse.bin novatek_ts-djn-NT36525-181116-08-channel.bin   
aw8624_rtp_silk.bin       novatek_ts-djn-NT36525B-190103-146-channel.bin 
cpp_firmware_v1_10_0.fw   wlan                                           
`
This is /vendor/firmware/ from stock working kernel, I can see novatk and ilitek. Can you please explain how to build them?

electimon, 2020-03-13:
@keyur2maru while on stock can you provide a lsmod? it should list the modules loaded.

keyur2maru, 2020-03-13:
@electimon 

`127|channel:/ $ lsmod
Module                  Size  Used by
wlan                 5279744  0 
fpc1020_mmi            16384  0 
aw8624                 49152  0 
drv2624_mmi            28672  0 
himax_mmi             151552  0 
ilitek_mmi           4263936  0 
nova_36525_mmi        233472  0 
tps61280               16384  0 
abov_sar               36864  0 
sx931x_triple          36864  0 
sx932x_sar             32768  0 
tas2560                36864  1 
tas2560_calib          16384  0 
mmi_sys_temp           16384  1 
tzlog_dump             16384  0 
mmi_info               24576  1 tzlog_dump
mmi_annotate           16384  2 tzlog_dump,mmi_info
sensors_class          16384  3 abov_sar,sx931x_triple,sx932x_sar
watchdogtest           16384  0 
utags                  28672  0 
exfat                 126976  0 
machine_ext_dlkm      110592  0 
machine_dlkm           86016  0 
tas2560_algo           28672  2 tas2560_calib,machine_dlkm
wcd9335_dlkm          262144  1 machine_ext_dlkm
cpe_lsm_dlkm           40960  0 
wcd_cpe_dlkm           77824  2 wcd9335_dlkm,cpe_lsm_dlkm
analog_cdc_dlkm       569344  2 machine_dlkm
digital_cdc_dlkm       49152  2 machine_dlkm
stub_dlkm              16384  1 
mbhc_dlkm              49152  2 wcd9335_dlkm,analog_cdc_dlkm
wsa881x_analog_dlkm    36864  1 machine_dlkm
wsa881x_dlkm           45056  2 machine_ext_dlkm,wsa881x_analog_dlkm
wcd9xxx_dlkm           57344  2 wcd9335_dlkm,analog_cdc_dlkm
wcd_core_dlkm         122880  8 machine_ext_dlkm,machine_dlkm,wcd9335_dlkm,wcd_cpe_dlkm,mbhc_dlkm,wsa881x_analog_dlkm,wsa881x_dlkm,wcd9xxx_dlkm
swr_ctrl_dlkm          32768  1 wcd9335_dlkm
swr_dlkm               24576  2 wsa881x_dlkm,swr_ctrl_dlkm
pinctrl_wcd_dlkm       16384  0 
native_dlkm           126976  0 
platform_dlkm        1323008  33 native_dlkm
usf_dlkm               49152  0 
q6_dlkm               770048  12 machine_ext_dlkm,machine_dlkm,tas2560_algo,wcd_cpe_dlkm,analog_cdc_dlkm,digital_cdc_dlkm,wcd9xxx_dlkm,native_dlkm,platform_dlkm,usf_dlkm
adsp_loader_dlkm       16384  0 
apr_dlkm              167936  7 wcd9335_dlkm,analog_cdc_dlkm,digital_cdc_dlkm,platform_dlkm,usf_dlkm,q6_dlkm,adsp_loader_dlkm
q6_notifier_dlkm       16384  2 analog_cdc_dlkm,apr_dlkm
channel:/ $ 
`
Here is it. Thank you, boss!

electimon, 2020-03-13:
@keyur2maru 
To determine which touch you use you can rmmod one of these
 himax_mmi 151552 0 ilitek_mmi 4263936 0 nova_36525_mmi
and if touch is gone you know thats the module your display uses.

keyur2maru, 2020-03-13:
@electimon, when I did rmmod himax_mmi, device rebooted, checked with lsmod, it still was listed there.
rmmod ilitek_mmi didn't do anything, touch kept working, phone stayed on
rmmod nova_36525 did the trick and touch stopped working. How can I build nova_36525 with my kernel and flash it in the OS?

electimon, 2020-03-13:
@keyur2maru request a branch for channel here
https://github.com/MotorolaMobilityLLC/motorola-kernel-modules

keyur2maru, 2020-03-13:
@electimon thank you but how can I build it?

charleseb, 2020-03-13:
Hi  @keyur2maru,

Please avoid to use branches, use the pushed tags instead: https://github.com/MotorolaMobilityLLC/motorola-kernel-modules/releases/tag/MMI-PPYS29.105-134-7

keyur2maru, 2020-03-13:
@charleseb yes I'm using the pushed tags for channel. I just got the nova_36525_mmi.ko file but I'm not sure how can I load it once it boots? because touch doesn't work I cannot use adb

electimon, 2020-03-13:
Sorry charleseb, @keyur2maru you can use a init script to insmod it on boot.

keyur2maru, 2020-03-13:
@electimon thanks a lot sorry just a noob tryna make it work

charleseb, 2020-03-13:
Hi @electimon,

No problem, I've just recommended the use of tags because we apply some tags to existing branches when the current commit is already there. So it's better to use the tags to get the right version/commit.

keyur2maru, 2020-03-13:
@electimon should I repack system.img with new .ko file and then insmod it in init ?

electimon, 2020-03-13:
@keyur2maru  If your modifying stock then the init script already exists to load the module. Putting it in /vendor/lib/modules/ replacing the old one should be enough.

charleseb, 2020-03-13:
Readme file: https://github.com/MotorolaMobilityLLC/readme/blob/master/MMI-PPYS29.105-134-7.txt

keyur2maru, 2020-03-13:
@charleseb can you update the readme for required modules too?

keyur2maru, 2020-03-13:
@electimon I've been trying to figure out how to repack vendor.img which was obtained from stock rom. Any lead?

charleseb, 2020-03-14:
Hi @keyur2maru,

Motorola Kernel Modules lines updated to https://github.com/MotorolaMobilityLLC/readme/blob/master/MMI-PPYS29.105-134-7.txt

keyur2maru, 2020-03-14:
@charleseb Thank you, sir!!

keyur2maru, 2020-03-15:
@charleseb 
With the new instructions, I'm facing this error 
Cannot use CONFIG_CC_STACKPROTECTOR_STRONG: -fstack-protector-strong not supported by compiler
Should I disable it from the config?

keyur2maru, 2020-03-16:
Vl53L0 Module Build Error
`kd-pax0r@ubuntu:~$ make -j24 -C kernel/msm-4.9 M=../../motorola/kernel/modules/drivers/input/misc/vl53L0 O=$kernel_out_dir ARCH=arm CROSS_COMPILE=$ccross KCFLAGS=-mno-android modules
make: Entering directory '/home/kd-pax0r/kernel/msm-4.9'
make[1]: Entering directory '/home/kd-pax0r/out/target/product/generic/obj/kernel/msm-4.9'
  CC [M]  ../../motorola/kernel/modules/drivers/input/misc/vl53L0/stmvl53l0_module-i2c.o
  CC [M]  ../../motorola/kernel/modules/drivers/input/misc/vl53L0/stmvl53l0_module-cci.o
/home/kd-pax0r/kernel/msm-4.9/../../motorola/kernel/modules/drivers/input/misc/vl53L0/stmvl53l0_module-cci.c: In function 'stmvl53l0_cci_init':
/home/kd-pax0r/kernel/msm-4.9/../../motorola/kernel/modules/drivers/input/misc/vl53L0/stmvl53l0_module-cci.c:289:3: error: implicit declaration of function 'media_entity_init' [-Werror=implicit-function-declaration]
   media_entity_init(&data->msm_sd.sd.entity, 0, NULL, 0);
   ^
/home/kd-pax0r/kernel/msm-4.9/../../motorola/kernel/modules/drivers/input/misc/vl53L0/stmvl53l0_module-cci.c:290:25: error: 'struct media_entity' has no member named 'type'
   data->msm_sd.sd.entity.type = MEDIA_ENT_T_V4L2_SUBDEV;
                         ^
/home/kd-pax0r/kernel/msm-4.9/../../motorola/kernel/modules/drivers/input/misc/vl53L0/stmvl53l0_module-cci.c:290:33: error: 'MEDIA_ENT_T_V4L2_SUBDEV' undeclared (first use in this function)
   data->msm_sd.sd.entity.type = MEDIA_ENT_T_V4L2_SUBDEV;
                                 ^
/home/kd-pax0r/kernel/msm-4.9/../../motorola/kernel/modules/drivers/input/misc/vl53L0/stmvl53l0_module-cci.c:290:33: note: each undeclared identifier is reported only once for each function it appears in
/home/kd-pax0r/kernel/msm-4.9/../../motorola/kernel/modules/drivers/input/misc/vl53L0/stmvl53l0_module-cci.c:291:25: error: 'struct media_entity' has no member named 'group_id'
   data->msm_sd.sd.entity.group_id = MSM_CAMERA_SUBDEV_TOF;
                         ^
cc1: some warnings being treated as errors
/home/kd-pax0r/kernel/msm-4.9/scripts/Makefile.build:339: recipe for target '../../motorola/kernel/modules/drivers/input/misc/vl53L0/stmvl53l0_module-cci.o' failed
make[2]: *** [../../motorola/kernel/modules/drivers/input/misc/vl53L0/stmvl53l0_module-cci.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/home/kd-pax0r/kernel/msm-4.9/../../motorola/kernel/modules/drivers/input/misc/vl53L0/stmvl53l0_module-i2c.c:322:2: error: unknown field 'resume' specified in initializer
  .resume = stmv153l0_resume,
  ^
/home/kd-pax0r/kernel/msm-4.9/../../motorola/kernel/modules/drivers/input/misc/vl53L0/stmvl53l0_module-i2c.c:322:2: warning: initialization from incompatible pointer type
error, forbidden warning: stmvl53l0_module-i2c.c:322
/home/kd-pax0r/kernel/msm-4.9/scripts/Makefile.build:339: recipe for target '../../motorola/kernel/modules/drivers/input/misc/vl53L0/stmvl53l0_module-i2c.o' failed
make[2]: *** [../../motorola/kernel/modules/drivers/input/misc/vl53L0/stmvl53l0_module-i2c.o] Error 1
/home/kd-pax0r/kernel/msm-4.9/Makefile:1613: recipe for target '_module_../../motorola/kernel/modules/drivers/input/misc/vl53L0' failed
make[1]: *** [_module_../../motorola/kernel/modules/drivers/input/misc/vl53L0] Error 2
make[1]: Leaving directory '/home/kd-pax0r/out/target/product/generic/obj/kernel/msm-4.9'
Makefile:152: recipe for target 'sub-make' failed
make: *** [sub-make] Error 2
make: Leaving directory '/home/kd-pax0r/kernel/msm-4.9'
`

Can someone please help me debug this?

awheeler-mmi, 2020-03-18:
Do you have the correct motorola/kernel tag checked out?  I wouldn't expect the correct tag to have that driver.

vinimed, 2022-03-11:
Hello @keyur2maru, due to the long time without updates, we are closing the issue. If you are still facing errors, please re-open the issue. 
Thank you in advance.

closed 2022-03-11 by vinimed