MotorolaMobilityLLC/readme #45: [REQUEST] Build instructions for XT2225-2(rhodep-cn)

HorseLuke, 2023-01-18:
Hello, could you please release build instructions for XT2225-2(rhodep-cn) S1SU32.73-34-7 and S1SU32.73-112-22? Thanks!

PS: I found kernel source S1SU32.73-34-7(https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-S1SU32.73-34-7 ), but could not found kernel source S1SU32.73-112-22.

HorseLuke, 2023-02-08:
OK, I found a solution.

For those who are interested, just follow build instructions for "The Generic Kernel Image (GKI) project". XT2225-2(rhodep-cn) is compatible with "common-android11-5.4" branch.

Links: https://source.android.com/docs/setup/build/building-kernels

closed 2023-02-08 by HorseLuke

chenyt9, 2023-02-08:
@HorseLuke ,pls refer to :https://github.com/MotorolaMobilityLLC/readme/blob/master/MMI-S1SU32.73-34-7.txt

HorseLuke, 2023-02-08:
@chenyt9 thanks!