MotorolaMobilityLLC/readme #46: [REQUEST] Build instructions for Moto g200 5G (xpeng)

ThEMarD, 2023-01-24:
Heyyo, I am requesting readme build instructions for the Moto g200 5G (xpeng).

I see that tag MMI-S1RXC32.50-37-2 is released so build instructions for this or a newer release tag would be appreciated.

Thank you,
Marc Bourgoin

chenyt9, 2023-02-08:
@ThEMarD ,pls refer to :https://github.com/MotorolaMobilityLLC/readme/blob/master/MMI-S1RXC32.50-37-2.txt

closed 2023-02-08 by chenyt9

rogers2602, 2023-03-13:
@chenyt9 I tried to compile following the instructions in the readme that was released for xpeng, but I always get this error and I don't know what else to do, could you give me some light?

even the instructions there point to google's git directories of binaries that don't even exist anymore, to solve this I needed to do some git reverts to adapt the instructions.

the error i am having:

make[3]: *** No rule to process target 'arch/arm64/boot/dts/allwinner/sun50i-a64-amarula-relic.dtb', needed by '__build'. Stop.
make[2]: *** [/home/rogers/Documents/compilation/my_top_dir/kernel/msm-5.4/scripts/Makefile.build:558: arch/arm64/boot/dts/allwinner] Error 2
make[2]: ** Waiting for other processes to terminate.
make[1]: *** [/home/rogers/Documents/compilation/my_top_dir/kernel/msm-5.4/Makefile:1381: dtbs] Error 2
make[1]: ** Waiting for other processes to terminate.
HOSTLD scripts/genksyms/genksyms
make[1]: Leaving directory '/home/rogers/Documentos/compilacao/my_top_dir/out/target/product/generic/obj/kernel/msm-5.4'
make: *** [Makefile:183: sub-make] Error 2
make: Leaving directory '/home/rogers/Documentos/compilacao/my_top_dir/kernel/msm-5.4'