MotorolaMobilityLLC/readme #47: Build instructions for motorola edge 30 fusion (tundra)

schqiushui, 2023-02-02:
Please update building instructions for tundra, It‘s hard to build a kernel whitout extra kernel moduls config for this device. 
Thank you!

chenyt9, 2023-02-08:
@schqiushui ,Pls refer to;https://github.com/MotorolaMobilityLLC/readme/blob/master/MMI-S3SJ32.1-86-1.txt

closed 2023-02-08 by chenyt9

schqiushui, 2023-02-12:
Thank you very much for your updated document, but I found a problem that the kernel modules configurations are missing in that file. Compared with xpeng, another msm-5.4 kernel‘s building instruction, there are no configurations of WLAN drivers and other kernel modules. So, am I missing something? 
please help me. 
Thanks again.

chenyt9, 2023-02-13:
@schqiushui ， Aleady update the readme file. Pls see the file for detail. Thanks.