MotorolaMobilityLLC/readme #48: Drvgen.py

damnthefall, 2023-02-10:
@damnthefall ,Pls refer to:https://github.com/MotorolaMobilityLLC/readme/blob/master/MMI-S1SA32.47-77

_Originally posted by @chenyt9 in https://github.com/MotorolaMobilityLLC/readme/issues/41#issuecomment-1421911586_

@chenyt9 Thank you for posting the build instructions for this device, however where can I find Drvgen.py? I am unable to follow the build instructions due to not knowing where to get this file....

chenyt9, 2023-02-20:
@damnthefall Pls refer to：https://github.com/MotorolaMobilityLLC/kernel-mtk/blob/android-12-release-ssa/DrvGen.py

closed 2023-02-20 by chenyt9

damnthefall, 2023-02-20:
@chenyt9 Thank you for posting the DrvGen.py file, however it is useless without the entire kernel-mtk/tools/dct folder that contains the python objects and other configuration files required to properly import the DrvGen python module. Also I noticed in commit a39e92d6a245d885a9e729a8186e2605ad286643 of kernel-mtk DrvGen and Dct have been completely removed from the build process by Mediatek. I think I can pull the required files to make DrvGen.py work from the commit messages, but since this is obviously not the proper way of building the MTK kernel anymore  is there any chance you have the updated build instructions?

chenyt9, 2023-02-21:
@damnthefall ，
I have created two dtsi file and uploaded：
https://github.com/MotorolaMobilityLLC/kernel-mtk/commit/e1c4528cfb005066e1077f1f241de5a67665bc89

And also upload the readme：
 https://github.com/MotorolaMobilityLLC/readme/blob/master/MMI-S1SA32.47-77
 pls copy it as following：
cp kernel-4.14/dvt2_cust.dtsi $kernel_out_dir/arch/arm64/boot/dts/austin_dvt2/cust.dtsi
cp kernel-4.14/evb_cust.dtsi $kernel_out_dir/arch/arm64/boot/dts/austin_evb/cust.dtsi