MotorolaMobilityLLC/readme #50: [Request] Moto G20 (JAVA) (RTAS31.68-66-2)

Claravallac, 2023-02-22:
Hi, we are at a loss to develop anything for the G20, could you please release the kernel?

Claravallac, 2023-03-08:
Thanks for release!

https://github.com/MotorolaMobilityLLC/readme/blob/master/MMI-RTA31.68-29

But, Could you send the complete instructions and make all the commits available?

Claravallac, 2023-03-08:
@TinnoOSS

shadzin, 2023-03-09:
Source kernel RTAS31.68-66.2 pls @TinnoOSS

fuyukihidekii, 2023-03-09:
https://github.com/MotorolaMobilityLLC/kernel-sprd/issues/6#issue-1616388329
@TinnoOSS

TinnoOSS, 2023-03-21:
@ALL,version released:
https://github.com/MotorolaMobilityLLC/kernel-sprd/releases/tag/MMI-RTAS31.68-66-2

Claravallac, 2023-03-21:
@TinnoOSS The instructions are incomplete bro

TinnoOSS, 2023-04-03:
hi Claravallac,
Excuse me, where is it incomplete? Can you describe it more clearly?

Claravallac, 2023-04-03:
@TinnoOSS The instruction lines are not for compiling the kernel, 15 lines without specifications

Claravallac, 2023-04-03:
@TinnoOSS And some developers have passed on that the kernel is smaller than it should be

Claravallac, 2023-04-03:
https://github.com/MotorolaMobilityLLC/readme/issues/53#issue-1640530106