MotorolaMobilityLLC/readme #53: [REQUEST] Functional Build Instructions Moto G20 [RTAS31.68-66-3]

Claravallac, 2023-03-25:
Please review the moto G20 (Java) kernel and build instructions that you released a little while ago.

The instructions only have 15 lines and the kernel weighs no more than 300 MB

@TinnoOSS @chenyt9 

[Incomplete]
https://github.com/MotorolaMobilityLLC/kernel-sprd/releases/tag/MMI-RTAS31.68-66-2

[Incomplete]
https://github.com/MotorolaMobilityLLC/readme/blob/master/MMI-RTAS31.68-66-2

Claravallac, 2023-03-27:
@asbraga-motorola

Claravallac, 2023-04-03:
@TinnoOSS

Claravallac, 2023-04-03:
Wrong

TinnoOSS, 2023-04-04:
hi Claravallac，
Our team downloads github open source kernel code and compiled script, and the local complete project code is carried out for verification, we did not find any problems, the kernel package and compiled script are completely identical to the local one, there is no loss. Do you need it please?

fuyukihidekii, 2023-04-06:
@TinnoOSS It would be nice to have access to tools and instructions specific to Unisoc. The fact that using U-BOOT makes things more complicated than a device with MBM.

If possible, could you provide information on which repos we need to have the kernel 100% functional? Like, which repos are the headers, modules etc.

- hideki

dsk46, 2023-04-11:
El fracaso se espera y en cierta forma se desea, pues elimina la angustia de esperanzarse en algo que a fin de cuentas no es posible y libera

TinnoOSS, 2023-04-17:
hi @fuyukihidekii ,
The build instructions have been updated, please go to the link below to view, if you have any questions, please feel free to consult.
https://github.com/MotorolaMobilityLLC/readme/blob/master/MMI-RTAS31.68-66-2

fuyukihidekii, 2023-06-23:
@TinnoOSS apologies for taking a long time to reply.

I can confirm is all okay and reproducible with updated instructions.

Joao-Victor-Liporini, 2023-09-28:
Hi @TinnoOSS, can you update the build instructions from the last kernel update 68-66-3?
because the instructions are for the old version (68-66-2) and no longer work in the new version (68-66-3)
If you can do this, not only me but the entire Moto G20 community would be immensely grateful.

Thanks in advance ~ João Victor

fuyukihidekii, 2023-09-28:
> Hi @TinnoOSS, can you update the build instructions from the last kernel update 68-66-3? because the instructions are for the old version (68-66-2) and no longer work in the new version (68-66-3) If [...]

> Hi @TinnoOSS, can you update the build instructions from the last kernel update 68-66-3? because the instructions are for the old version (68-66-2) and no longer work in the new version (68-66-3) If [...]

Did they didn't released 8-66-3 source yet, and everything is fine.

Remove everything between line 426 - 431 on main Makefile and you will get the final build product without any newer failures.

https://github.com/MotorolaMobilityLLC/kernel-sprd/blob/7ab39b1706bae38466f87acf01f4eb4c1b8cdd37/Makefile#L426

Joao-Victor-Liporini, 2023-09-28:
> > Hi @TinnoOSS, can you update the build instructions from the last kernel update 68-66-3? because the instructions are for the old version (68-66-2) and no longer work in the new version (68-66-3) If you can do this, not only me but the entire Moto G20 community would be immensely grateful.
> > 
> > Thanks in advance ~ João Victor
> 
> 
> 
> > Hi @TinnoOSS, can you update the build instructions from the last kernel update 68-66-3? because the instructions are for the old version (68-66-2) and no longer work in the new version (68-66-3) If you can do this, not only me but the entire Moto G20 community would be immensely grateful.
> > 
> > Thanks in advance ~ João Victor
> 
> Did they didn't released 8-66-3 source yet, and [EDIT: 'everything' with the current kernel source ] is fine.
> 
> Remove everything between line 426 - 431 on main Makefile and you will get the final build product without any newer failures.
> 
> https://github.com/MotorolaMobilityLLC/kernel-sprd/blob/7ab39b1706bae38466f87acf01f4eb4c1b8cdd37/Makefile#L426

Ok, I'll try