MotorolaMobilityLLC/readme #57: [REQUEST] Build instructions for Motorola Edge 20 Pro (pstar) T1RA33.55-15-10 (XT2153-1)

richard4O4, 2023-05-26:
Could you please share the instructions for how to build the [MMI-T1RA33.55-15-10](https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-T1RA33.55-15-10) kernel for pstar? We would be very appreciated if you can.

chenyt9, 2023-05-26:
@richard4O4 ,
Pls refer to https://github.com/MotorolaMobilityLLC/readme/blob/master/MMI-T1RA33.55-15-10
Thanks.

closed 2023-05-26 by chenyt9

richard4O4, 2023-05-27:
> @richard4O4 , Pls refer to https://github.com/MotorolaMobilityLLC/readme/blob/master/MMI-T1RA33.55-15-10 Thanks.

@chenyt9 Thanks a lot. But I found the wrong instructions you posted, it's for Motorola edge **40** pro (codename: `rtwo`). Could you please correct it? I need the instructions for **Motorola Edge 20 Pro (pstar) T1RA33.55-15-10 (XT2153-1)**

Check these lines:

* [L1 ](https://github.com/MotorolaMobilityLLC/readme/blob/080647ac2458458b1814e2de9b2a9a1e24118c2e/MMI-T1RA33.55-15-10#LL1C45-L1C45)
* [L51 ](https://github.com/MotorolaMobilityLLC/readme/blob/080647ac2458458b1814e2de9b2a9a1e24118c2e/MMI-T1RA33.55-15-10#L51)
* [L91](https://github.com/MotorolaMobilityLLC/readme/blob/080647ac2458458b1814e2de9b2a9a1e24118c2e/MMI-T1RA33.55-15-10#L91)

```bash
MMI-T1RA33.55-15-10 For Motorola edge 40 pro

# ......

TARGET_PRODUCT="rtwo"

# ......

./msm-kernel/arch/arm64/configs/vendor/ext_config/moto-kalama-rtwo.config
```