MotorolaMobilityLLC/readme #65: Missing file in MMI-S3SG32.39-191R2 readme

dynamikmechanik, 2024-01-31:
Hi, @chenyt9 I am hitting an error on line 41 of the readme for `MMI-S3SG32.39-191R2`, 
```
$ python vendor/mediatek/proprietary/tools/dct/python/DrvGen.py kernel-4.19/drivers/misc/mediatek/dws/mt6765/maui_evb.dws $kernel_out_dir/arch/arm/boot/dts/maui_evb $kernel_out_dir/arch/arm/boot/dts/maui_evb cust_dtsi
python: can't open file '[my_top_dir]/vendor/mediatek/proprietary/tools/dct/python/DrvGen.py': [Errno 2] No such file or directory
```
I have tag `MMI-S3SG32.39-191R2` checked out of `kernel-mtk` and I do not see a `vendor` directory or `DrvGen.py`. Is there another step I need to take to include this directory? Thank you.

_Originally posted by @dynamikmechanik in https://github.com/MotorolaMobilityLLC/readme/issues/62#issuecomment-1919315774_

dynamikmechanik, 2024-01-31:
Also, what commit/tag should I check out for `arm-linux-androideabi-49` on line 19:
```
git clone https://android.googlesource.com/platform/prebuilts/gcc/linux-x86/arm/arm-linux-androideabi-4.9
```
Do I need to checkout a specific commit/tag for the build-tools? Thank you.

dynamikmechanik, 2024-01-31:
Should I run the following to get clan prebuilts?
```
mkdir -vp $my_top_dir/prebuilts/clang/host
cd $my_top_dir/prebuilts/clang/host
git clone https://android.googlesource.com/platform/prebuilts/clang/host/linux-x86
```
If so, what commit/tag should I use? Thank you.

chenyt9, 2024-02-01:
@dynamikmechanik ,
I have updated the readme, the command is to generate cust_dtsi file  could you get the file from https://github.com/MotorolaMobilityLLC/kernel-mtk/blob/MMI-S3SG32.39-191R2/cust.dtsi
mv cust_dtsi $kernel_out_dir/arch/arm/boot/dts/maui_evb/.
Thanks.

chenyt9, 2024-02-01:
@dynamikmechanik ,
https://android.googlesource.com/platform/prebuilts/gcc/linux-x86/arm/arm-linux-androideabi-4.9/+/refs/tags/android-12.0.0_r11

chenyt9, 2024-02-01:
@dynamikmechanik 
https://android.googlesource.com/platform/prebuilts/clang/host/linux-x86/+/refs/heads/android12-qpr1-release

dynamikmechanik, 2024-02-02:
Thanks for the responses. Now I am hitting an error from line 59 to the end of the readme. Each command errors with `No such file or directory` as it cannot find the Makefile.