MotorolaMobilityLLC/readme #66: [REQUEST] Build instructions for Edge 30 Neo (XT2245-1) - T1SSMS33.1-121-4-8

juergh, 2024-02-08:
That and a pointer to updated kernel sources would be really helpful. Thanks a lot!

juergh, 2024-02-11:
Ok, I think I figured out how to build the kernel based on information from the other build instruction files. But my device is currently running `Linux localhost 5.4.233-moto-gb821dcf983ba #1 SMP PREEMPT Thu Dec 21 07:43:02 CST 2023 aarch64 Toybox` and I can't find `b821dcf983ba` in https://github.com/MotorolaMobilityLLC/kernel-msm. Would it be possible to update the `android-13-release-t1ssms33.1-121-4-5` branch to `-8` or whatever is current?

Thanks!

juergh, 2024-05-06:
I see there's a tag `MMI-T1SSMS33.1-121-4-8` now, so yes. Thanks!

closed 2024-05-06 by juergh

AMT-Cheif, 2024-05-07:
> possible to update the `android-13-release-t1ssms33.1-121-4-5` branch to `-8` or whatever is curren

> I see there's a tag `MMI-T1SSMS33.1-121-4-8` now, so yes. Thanks!
I'm on my first build, can I contact you? Can you assist me in this journey?